immortalwrt/luci #62: master 分支的 filebrowser，在 luci 不能启动

fanck0605, 2021-07-15:
[image]

dalao，master 分支的 filebrowser 在 luci 不能启动。
手动执行 `/etc/init.d/filebrowser reload` 才行。
在脚本里加了点标记，在 luci 选择已启用时，`/etc/init.d/filebrowser` 的确没被执行
```
#!/bin/sh /etc/rc.common
# Copyright (C) 2019 [CTCGFW] Project OpenWRT

echo "exec_script" >> /fb_test

. /lib/functions.sh
. /lib/functions/procd.sh

USE_PROCD=1

START=90
STOP=10

init_conf() {
        echo "init_conf" >> /fb_test
        config_load "filebrowser"
        config_get "addr_type" "config" "addr_type" "lan"
        ...
```

dfldylan, 2022-07-19:
> 这个东西的默认用户名密码是啥啊……

哦admin

himan85, 2022-11-16:
居然有人提到这个问题，2022年最新版本貌似还没解决

msylgj, 2022-11-16:
luci没有问题, 更换用immortalwrt/packages的filebrowser包即可. https://github.com/immortalwrt/packages/commit/3554d0b6d48089c799a39adf669f5218f7b21fe8

closed 2022-11-16 by msylgj