immortalwrt/luci #80: [18.06-K5.4]无法启动docker daemon

ShoukakuChuuCC, 2021-09-17:
问题描述
luci-app-dockerman无法启动，提示：Please input the SOCKET PATH of docker daemon!

主机名 | ImmortalWrt
型号 | Raspberry Pi 4 Model B Rev 1.1
架构 | ARMv8 Processor rev 3 (v8l) x 4 (1500Mhz, 40.4°C)
固件版本 | ImmortalWrt 18.06-SNAPSHOT r10862-6b3a9d3a3 / LuCI openwrt-18.06-k5.4 branch (git-21.256.04965-a0664ec)
内核版本 | 5.10.64

[image]
[image]

1715173329, 2021-09-17:
I don't use docker on openwrt, can you take a look?
Ping: @EnnawYang @SuLingGG

EnnawYang, 2021-09-18:
上游的dockerman其实已经解决了不能启动的问题，但我试过bridge还是不通的，不知道是哪里的问题（

1715173329, 2021-09-22:
updated.

ShoukakuChuuCC, 2021-09-22:
更新后还是无法启动docker daemon
[image]
[image]

ShoukakuChuuCC, 2021-09-22:
配置页面直接一点信息都没有了

1715173329, 2021-10-05:
This issue should be addressed.
@ShoukakuChuuCC Can you make a test?

ShoukakuChuuCC, 2021-10-06:
经验证，已修复，感谢大佬

closed 2021-10-06 by ShoukakuChuuCC