immortalwrt/luci #211: openvpn client /luci 集成

evenlee, 2022-09-11:
请问能集成openvpn client和对应的可以直接上传openvpn 配置文件的luci app  package吗？

1715173329, 2022-09-11:
idk what it is

evenlee, 2022-09-11:
[openvpn-openssl](https://openwrt.org/packages/pkgdata/openvpn-openssl) and [luci-app-openvpn](https://openwrt.org/packages/pkgdata/luci-app-openvpn)

I want to use openvpn client in a router, thanks

1715173329, 2022-09-11:
it's upstreamed, so we have it too
idk what do u mean by "integrate", it's already available in our opkg feeds

evenlee, 2022-09-11:
I am using [SuLingGG](https://github.com/SuLingGG) /[OpenWrt-Rpi](https://github.com/SuLingGG/OpenWrt-Rpi), which is based on immortalwrt and seems does not have openvpn client installed by default, thanks for confirming.

closed 2022-09-11 by evenlee