immortalwrt/luci #217: For a resolution refer to Documentation/kbuild/kconfig-language.rst subsection "Kconfig recursive dependency limitations"

netlovehf, 2022-10-22:
/immortalwrt$ make defconfig
tmp/.config-package.in:87104:error: recursive dependency detected!
tmp/.config-package.in:87104:  symbol PACKAGE_iptables is selected by PACKAGE_luci-app-passwall_Iptables_Transparent_Proxy
tmp/.config-package.in:65395:  symbol PACKAGE_luci-app-passwall_Iptables_Transparent_Proxy default is visible depending on PACKAGE_firewall4
tmp/.config-package.in:326:  symbol PACKAGE_firewall4 depends on PACKAGE_kmod-ipt-nat
tmp/.config-package.in:15014:  symbol PACKAGE_kmod-ipt-nat is selected by PACKAGE_kmod-ipt-dnetmap
tmp/.config-package.in:14797:  symbol PACKAGE_kmod-ipt-dnetmap is selected by PACKAGE_iptables-mod-dnetmap
tmp/.config-package.in:87348:  symbol PACKAGE_iptables-mod-dnetmap depends on PACKAGE_iptables
For a resolution refer to Documentation/kbuild/kconfig-language.rst
subsection "Kconfig recursive dependency limitations"

#
# No change to .config

[image]

1715173329, 2022-10-23:
I don't see this issue. Which branch are you on?

netlovehf, 2022-10-25:
> I don't see this issue. Which branch are you on?

openwrt-18.06-k5.4

1715173329, 2022-10-25:
fixed

1715173329, 2022-10-25:
18.06-k5.4 is EoL. please consider upgrading to 21.02.

closed 2022-10-25 by 1715173329