immortalwrt/luci #236: netdata编译了 没显示出来 用不了

yyhhkya, 2023-01-20:


zcluo, 2023-02-22:
别编了一自己下一个安装下
if [ ! -f "/opt/netdata/bin/netdata" ];then
	mkdir -p /opt/netdata/etc/netdata
	cp -f /etc/netdata/*.conf /opt/netdata/etc/netdata/
##	cd /root/netdata-offline
##	chmod +x install.sh
    wget -O /tmp/netdata-kickstart.sh https://my-netdata.io/kickstart.sh && sh /tmp/netdata-kickstart.sh --stable-channel --prepare-offline-install-source  /tmp/netdata-offline
	cd /tmp/netdata-offline
	nohup ./install.sh --non-interactive &
else
	sleep 5
	[  -d "/tmp/netdata-offline/" ] && rm -rf /tmp/netdata-offline
	ps -ef | grep netdata | grep -v grep | awk {'print $1'} | xargs kill -9
	. /opt/netdata/bin/netdata -P /tmp/netdata.pid >> /tmp/netdata.out 2>&1 &
fi

zcluo, 2023-02-22:
自己下一个安装。包里的版本太低了
`if [ ! -f "/opt/netdata/bin/netdata" ];then
	mkdir -p /opt/netdata/etc/netdata
	cp -f /etc/netdata/*.conf /opt/netdata/etc/netdata/
##	cd /root/netdata-offline
##	chmod +x install.sh
    wget -O /tmp/netdata-kickstart.sh https://my-netdata.io/kickstart.sh && sh /tmp/netdata-kickstart.sh --stable-channel --prepare-offline-install-source  /tmp/netdata-offline
	cd /tmp/netdata-offline
	nohup ./install.sh --non-interactive &
else
	sleep 5
	[  -d "/tmp/netdata-offline/" ] && rm -rf /tmp/netdata-offline
	ps -ef | grep netdata | grep -v grep | awk {'print $1'} | xargs kill -9
	. /opt/netdata/bin/netdata -P /tmp/netdata.pid >> /tmp/netdata.out 2>&1 &
fi`

zcluo, 2023-02-22:
> if [ ! -f "/opt/netdata/bin/netdata" ];then
	mkdir -p /opt/netdata/etc/netdata
	cp -f /etc/netdata/*.conf /opt/netdata/etc/netdata/
##	cd /root/netdata-offline
##	chmod +x install.sh
    wget -O /tmp/netdata-kickstart.sh https://my-netdata.io/kickstart.sh && sh /tmp/netdata-kickstart.sh --stable-channel --prepare-offline-install-source  /tmp/netdata-offline
	cd /tmp/netdata-offline
	nohup ./install.sh --non-interactive &
else
	sleep 5
	[  -d "/tmp/netdata-offline/" ] && rm -rf /tmp/netdata-offline
	ps -ef | grep netdata | grep -v grep | awk {'print $1'} | xargs kill -9
	. /opt/netdata/bin/netdata -P /tmp/netdata.pid >> /tmp/netdata.out 2>&1 &
fi

zcluo, 2023-02-22:
`if [ ! -f "/opt/netdata/bin/netdata" ];then
	mkdir -p /opt/netdata/etc/netdata
	cp -f /etc/netdata/*.conf /opt/netdata/etc/netdata/
    wget -O /tmp/netdata-kickstart.sh https://my-netdata.io/kickstart.sh && sh /tmp/netdata-kickstart.sh --stable-channel --prepare-offline-install-source  /tmp/netdata-offline
	cd /tmp/netdata-offline
	nohup ./install.sh --non-interactive &
else
	sleep 5
	[  -d "/tmp/netdata-offline/" ] && rm -rf /tmp/netdata-offline
	ps -ef | grep netdata | grep -v grep | awk {'print $1'} | xargs kill -9
	. /opt/netdata/bin/netdata -P /tmp/netdata.pid >> /tmp/netdata.out 2>&1 &
fi`

1715173329, 2024-09-19:
需要 netdata 处于运行状态才会显示。此外需要清理 luci 缓存。

closed 2024-09-19 by 1715173329