immortalwrt/luci #290: luci-app-frpc和luci-app-frps在21.02中不能正常运行，没有总开关

gdzhujun, 2023-03-23:
### Describe the bug

luci-app-frpc和luci-app-frps在21.02中不能正常运行，没有总开关，对比18.06版本是正常的，尝试过把18.06中的luci-app-frpc和luci-app-frps复制到21.02中编译，开关是出来了，但还是无法正常使用

### ImmortalWrt version

21.02

### ImmortalWrt target/subtarget

X86

### Device

X86

### Image kind

Official downloaded image

### Terms

- [X] I am reporting an issue for ImmortalWrt, not an unsupported fork.

1715173329, 2023-03-23:
目前设计就是这么个设计，上游没加 enabled 开关，如有需求请给上游提交 issue。

closed 2023-03-23 by 1715173329

gdzhujun, 2023-06-19:
能否共享一下

gdzhujun, 2023-06-20:
老哥邮箱发一份吧 gdzhujun1@gmail.com  谢谢啦

zxlhhyccc, 2023-06-22:
> 老哥邮箱发一份吧 [gdzhujun1@gmail.com](mailto:gdzhujun1@gmail.com) 谢谢啦

明天发你！

zxlhhyccc, 2023-06-23:
> 老哥邮箱发一份吧 [gdzhujun1@gmail.com](mailto:gdzhujun1@gmail.com) 谢谢啦

你去下载吧（luci-app-frps的日志未按照luci-app-frpc的方式优化）：
````
svn export https://github.com/zxlhhyccc/bf-package-master/trunk/lean/luci-app-frpc
````
````
svn export https://github.com/zxlhhyccc/bf-package-master/trunk/lean/luci-app-frps
````