immortalwrt/luci #363: luci-app-zerotier 不应将运行时文件存放在 `/etc` 中

CrazyBoyFeng, 2023-11-13:
zerotier 是用来内网穿透的工具，应该不用过多介绍。

在使用 luci-app-zerotier 的过程中，我发现它将运行时文件存放在 `/etc/config/zero` 中，导致 flash 被频繁读写、大量占用空间。
研究代码发现是这两句将运行时文件的存储位置定义在了 `/etc/config/zero` 中：
https://github.com/immortalwrt/luci/blob/1f8af453103ba07e0337f15e5f0488d4d92cf2bf/applications/luci-app-zerotier/root/etc/init.d/zerotier#L30-L31

这里的 `$config_path` 变量虽然名为配置，但实际存储的内容主要为：各节点信息、与各节点通信的密钥、统计数据、PID 进程锁、PORT 端口锁。  
这其中的大多数据都是变动的，有的还会不断增加。因此并不适合持久化存储，尤其是在低配置的路由器上。  
[zerotier 官方软件包的服务项文件](https://github.com/openwrt/packages/blob/d91e43e2215b604c6175c4916cc488abd02bdc65/net/zerotier/files/etc/init.d/zerotier)默认是将它们存放在 `/var/lib/zerotier-one` 里，也就是存储在 RAM 里。  

我追溯了一下这段代码，最早是 lean 添加上去的。我没有找到他做这个改动的原因，也许是为了调试？我也没有在代码中找到别处调用了这个路径。

此外这个服务项的另一个问题就是初始化服务时没有清理旧的运行时文件。
zerotier 官方[在初始化时清理了 `/var/lib/zerotier-one_sample_config`](https://github.com/openwrt/packages/blob/d91e43e2215b604c6175c4916cc488abd02bdc65/net/zerotier/files/etc/init.d/zerotier#L31-L35)。

我认为应当参考 [zerotier 官方软件包的服务项文件](https://github.com/openwrt/packages/blob/d91e43e2215b604c6175c4916cc488abd02bdc65/net/zerotier/files/etc/init.d/zerotier) 将这两句代码换为
```shell
config_path=${CONFIG_PATH}_$cfg  # 将运行时文件路径定义为 /var/lib/zerotier-one_sample_config
rm -rf $config_path  # 清理旧的运行时文件
mkdir -p $config_path  # 创建运行时文件目录
```

其实最好是将整个`start_instance()` 函数换成 zerotier 官方软件包的代码。那样不仅上述问题能够得到解决，还能够支持用户在 `/etc/config/zerotier` 配置文件中自定义运行时文件的路径、端口等设置。

1715173329, 2024-09-19:
这是预期行为，否则配置文件、设备信息、密钥等重启之后直接丢失。不知道哪种规模的 zerotier 网络可以造成“频繁读写”与“大量占用空间”。

closed 2024-09-19 by 1715173329

CrazyBoyFeng, 2024-09-19:
“预期行为”这结论令人意外。  
我之所以认为这是个问题，是因为我有一个 flash 只有 16mb 的 yk-k1 路由运行了 `luci-app-zerotier` 半年之后 flash 被占满导致了这个问题暴露的。  
之前使用 openwrt 官库里的包没有遇到这个问题。是因为那个版本中的运行时文件并不存放在 flash 中，而是存放在 RAM（`/var/`）中，并且每次重启 zerotier 都会清理之前的运行时文件。  
据我检索，官方 zerotier 代码里仅在此一处清理了运行时文件。如果不设置这个清理，那么 zerotier 将不会删除任何运行时文件。  
关于我运行的 zerotier 网络的规模，可能确实比一般的用户要大一些。我参加了数个用于游戏联机和文件共享等功能的 public 网域。域内经常会有新旧用户进出。  
此外，删除 `/etc/config/zero/` 这个运行时目录并不会导致运行所必要的配置文件在重启之后丢失。真正用来存储 zerotier 静态配置（包含加入网络的 id 和 secret）的文件位于 `/etc/config/zerotier`，这也是 `luci-app-zerotier` 所操作的配置文件。`/etc/config/zero/` 里面并不是配置文件，而是运行时文件。它之所以在 `/etc/config/` 这个位置，是咱们的 `luci-app-zerotier` 从官方包定义的 `/var/lib/zerotier-one*/` 改过去的。官方包每次服务启动时都会删除这个目录。删除后，除了要与节点重新握手获得密钥会慢一点之外，在其它体验上并没有什么影响。