immortalwrt/luci #385: luci-app-openvpn-server 存在安全风险

xxxsen, 2024-01-22:
## Steps to reproduce:

这个包安装成功后，本身需要重建相关的证书, 但是`openvpncert.sh`这个脚本缺失了必要的函数, 例如`build-dh`, `build-key-server`等, 导致相关证书无法生成, 最终用了ipk包自身带的默认证书, 这个会导致一个问题, 从A服务器产生的ovpn文件可以用于连接B服务器。

https://github.com/immortalwrt/luci/blob/761f346e581f82530f2c8a464bef6bbb83482c32/applications/luci-app-openvpn-server/root/etc/openvpncert.sh#L6

## Actual behavior:

多个openvpn server使用了同一套证书/密钥

## Expected behavior:

不同的openvpn server产生不同的证书/密钥

## Additional Information:

测试的immortalwrt版本: 23.05.1
对应的ipk: https://downloads.immortalwrt.org/releases/23.05.1/packages/x86_64/luci/luci-app-openvpn-server_2.0_all.ipk

Vinke2, 2024-03-21:
@1715173329 这个安全问题很严重啊,还不解决吗

xlighting2017, 2024-03-21:
咋的，不解决 X日内罚款？

要不直接删掉，problem solved~

xxxsen, 2024-03-21:
> 咋的，不解决 X日内罚款？
> 
> 要不直接删掉，problem solved~

没时间处理的话，确实先删掉这个包合适一点，这个东西放着，别人下载了，风险还是挺大的。

xxxsen, 2024-03-21:
https://fofa.info/result?qbase64=InVkcCIgJiYgcG9ydCA9PTExOTQ%3D

像这种暴露出端口的，可以直接被扫出来，我不确定这个openvpn的包返回的数据里面有没有特殊的字符串序列可以用于匹配，如果有的话，那。。。

hakureiyuyuko, 2024-03-21:
[image]
这不是有吗？

Vinke2, 2024-03-21:
> ![Screenshot_2024-03-21-17-14-40-80_40deb401b9ffe8e1df2f1cc5ba480b12](https://private-user-images.githubusercontent.com/72778645/315225766-38ec2686-a187-4a4b-9829-9acd70bdafcc.jpg?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTEwMTI4MTcsIm5iZiI6MTcxMTAxMjUxNywicGF0aCI6Ii83Mjc3ODY0NS8zMTUyMjU3NjYtMzhlYzI2ODYtYTE4Ny00YTRiLTk4MjktOWFjZDcwYmRhZmNjLmpwZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDAzMjElMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwMzIxVDA5MTUxN1omWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPWEyYzE2MGYxMmIxZjA5MDMxNDUwN2MzYzJjOWU4ZTUwYTdiNGNiMTdlMGVkZjUwNzczNTNlZGU2YzBmNGI1NmEmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.Jf1ltZc6mV_puQFZP247eRF7n9-Ye3vYdoM_NPnnSY4) 这不是有吗？

咋用,现在编译出来的包都是这个默认的一模一样的证书

closed 2024-03-21 by 1715173329

imEgo, 2024-03-21:
```/etc/easy-rsa/vars```按照如下修改
```
# A little housekeeping: DO NOT EDIT THIS SECTION
#
# Easy-RSA 3.x does not source into the environment directly.
# Complain if a user tries to do this:
if [ -z "$EASYRSA_CALLER" ]; then
	echo "You appear to be sourcing an Easy-RSA *vars* file. This is" >&2
	echo "no longer necessary and is disallowed. See the section called" >&2
	echo "*How to use this file* near the top comments for more details." >&2
	return 1
fi

set_var EASYRSA			"/etc/easy-rsa"
set_var EASYRSA_OPENSSL		"openssl"
set_var EASYRSA_PKI		"$EASYRSA/pki"
set_var EASYRSA_TEMP_DIR	"$EASYRSA_PKI"
set_var EASYRSA_KEY_SIZE	1024
set_var EASYRSA_ALGO		rsa
set_var EASYRSA_CA_EXPIRE	3650
set_var EASYRSA_CERT_EXPIRE	3650
set_var EASYRSA_SSL_CONF	"$EASYRSA/openssl-easyrsa.cnf"
set_var EASYRSA_BATCH		"yes"
#set_var EASYRSA_DN		"cn_only"
set_var EASYRSA_REQ_COUNTRY	"US"
set_var EASYRSA_REQ_PROVINCE	"California"
set_var EASYRSA_REQ_CITY	"San Francisco"
set_var EASYRSA_REQ_ORG		"OpenWrt"
set_var EASYRSA_REQ_EMAIL	"openwrt@openwrt.org"
set_var EASYRSA_REQ_OU		"OpenWrt"
```

执行如下命令可以生成证书
```
echo -en "yes\n" | easyrsa init-pki
easyrsa gen-dh
easyrsa build-ca nopass
easyrsa build-server-full server nopass
easyrsa build-client-full client1 nopass
cp /etc/easy-rsa/pki/ca.crt /etc/openvpn/
cp /etc/easy-rsa/pki/dh.pem /etc/openvpn/dh.pem
cp /etc/easy-rsa/pki/issued/server.crt /etc/openvpn/
cp /etc/easy-rsa/pki/private/server.key /etc/openvpn/
cp /etc/easy-rsa/pki/issued/client1.crt /etc/openvpn/
cp /etc/easy-rsa/pki/private/client1.key /etc/openvpn/
```

目前遇到的问题是虽然指定了nopass，但是生成证书的时候还是要输入密码，尚不清楚是脚本nopass的原因还是openssl的原因

bozheng1988, 2024-03-23:
重构了 luci-app-openvpn-server

1. 适配openvpn-easy-rsa 3.1.3版本
2. 移除固件内置证书改为证书在首次启动时生成
3. 完善翻译
4. Web-UI增加重新生成证书按钮
5. 重构文件结构，脚本收敛至/etc/openvpn目录下

commit efb5a293 (2024-03-21): luci-app-openvpn-server: drop package

This application relies on the legacy openvpn-easy-rsa scripts to
generate certificates, which was removed in newer version.

For security reason, remove this package until someone migrates related
CA generation commands to fit new easyrsa utility.

Closes: #385
commit 327d1361 (2024-03-21): luci-app-openvpn-server: drop package

This application relies on the legacy openvpn-easy-rsa scripts to
generate certificates, which was removed in newer version.

For security reason, remove this package until someone migrates related
CA generation commands to fit new easyrsa utility.

Closes: #385

(cherry picked from commit efb5a2932e61c6deda21927b8b4db5f662449b33)