immortalwrt/luci #437: 请添加插件

142857X1, 2024-08-16:
mosdns
https://github.com/sbwml/luci-app-mosdns

app-unblockmusic
https://github.com/maxlicheng/luci-app-unblockmusic

luci-app-adguardhome
源码有知道的请补充
luci-app-easymesh
源码有知道的请补充
automount
源码有知道的请补充
ssrp源码
https://github.com/fw876/helloworld

nicholas-opensource, 2024-08-16:
mosdns：您愿意做maintainer吗？而且后端mosdns版本可能不一致。
UnblockNeteaseMusic：这个是最新的，而且天灵是主要贡献者，已内置。
https://github.com/UnblockNeteaseMusic/luci-app-unblockneteasemusic
AdguardHome：无存在意义，一个enable=1就可以运行，除非您愿意做maintainer，转写js并维护。
剩下三个在主线没有存在意义，尤其是使用iptables-nft会导致很多问题，猜测您估计从coolsnowwolf/lede转过来，两边是不一样的，希望可以多参考一下Openwrt官方文档。
FYI：
https://openwrt.org/docs/start

142857X1, 2024-08-17:
> mosdns：您愿意做maintainer吗？而且后端mosdns版本可能不一致。 UnblockNeteaseMusic：这个是最新的，而且天灵是主要贡献者，已内置。 https://github.com/UnblockNeteaseMusic/luci-app-unblockneteasemusic [...]

mosdns源码 sbwml 是维护者，
https://github.com/sbwml/luci-app-mosdns
https://github.com/QiuSimons/openwrt-mos
https://github.com/IrineSistiana/mosdns 

adgardhome 这个是内置的意思么改enable=1就能运行？

那个ssrp，是跟imm不兼容么？以前用的K2P一直在用ssrp，编译的是lienol源码 ，
这刚入手3000M Emmc才接触immortal ，
菜鸟一枚，并不能完全理解你讲的，

xlighting2017, 2024-08-17:
咱这里不支持点菜，用敬语点也不行；

想吃什么自己做，不会做可以学， 没学过也有 [FAQ文档](https://github.com/immortalwrt/user-FAQ)；

否则就等，等哪天掉下个饼，立马可以吃饱，具体哪天，可以等通知；

142857X1, 2024-08-17:
> 咱这里不支持点菜，用敬语点也不行；
> 
> 想吃什么自己做，不会做可以学， 没学过也有 [FAQ文档](https://github.com/immortalwrt/user-FAQ)；
> 
> 否则就等，等哪天掉下个饼，立马可以吃饱，具体哪天，可以等通知；

我本以为imm是类似openwrt.ai不用编译快速生成固件，因为我看到在生成固件的选项有预编译插件，就报告一些经常使用的而没在列表出现的插件，仅此而已。
首先我没点你，也没点你的菜，我对你也没用敬语！你用不着阴阳怪气的。

bradfordzhang, 2024-08-17:
> > 咱这里不支持点菜，用敬语点也不行；
> > 
> > 想吃什么自己做，不会做可以学， 没学过也有 [FAQ文档](https://github.com/immortalwrt/user-FAQ)；
> > 
> > 否则就等，等哪天掉下个饼，立马可以吃饱，具体哪天，可以等通知；
> 
> 我本以为imm是类似openwrt.ai不用编译快速生成固件，因为我看到在生成固件的选项有预编译插件，就报告一些经常使用的而没在列表出现的插件，仅此而已。
> 首先我没点你，也没点你的菜，我对你也没用敬语！你用不着阴阳怪气的。

https://fw.wrt.moe/
刷入系统后也可以使用opkg安装软件包。开源项目纯靠开发者为爱发电，你发个issus就让人家添加这个添加那个属实不太合适。

xlighting2017, 2024-08-17:
> > 咱这里不支持点菜，用敬语点也不行；
> > 想吃什么自己做，不会做可以学， 没学过也有 [FAQ文档](https://github.com/immortalwrt/user-FAQ)；
> > 否则就等，等哪天掉下个饼，立马可以吃饱，具体哪天，可以等通知；
> 
> 我本以为imm是类似openwrt.ai不用编译快速生成固件，因为我看到在生成固件的选项有预编译插件，就报告一些经常使用的而没在列表出现的插件，仅此而已。 首先我没点你，也没点你的菜，我对你也没用敬语！你用不着阴阳怪气的。

没说点“我”了，也没说点“我”的菜了， 怎么就强调起“我”来了呢；

开issue，写这么个标题， 把原本issue模板全删掉， 然后“就报告一...仅此而已”
-- 是不是（想）点菜，你可以想想再否认；

我说“咱”，是因为我也是小小的 contributor
只不过我不愿意 issue 变成点菜的地方， 有这种趋势也不行；

142857X1, 2024-08-17:
> > > 咱这里不支持点菜，用敬语点也不行；
> > > 想吃什么自己做，不会做可以学， 没学过也有 [FAQ文档](https://github.com/immortalwrt/user-FAQ)；
> > > 否则就等，等哪天掉下个饼，立马可以吃饱，具体哪天，可以等通知；
> > 
> > 
> > 我本以为imm是类似openwrt.ai不用编译快速生成固件，因为我看到在生成固件的选项有预编译插件，就报告一些经常使用的而没在列表出现的插件，仅此而已。
> > 首先我没点你，也没点你的菜，我对你也没用敬语！你用不着阴阳怪气的。
> 
> https://fw.wrt.moe/ 刷入系统后也可以使用opkg安装软件包。开源项目纯靠开发者为爱发电，你发个issus就让人家添加这个添加那个属实不太合适。 而且mosdns和adguard这种完全没有luci存在必要的软件，在config文件启用即可 至于ssrp的问题，nicholas大佬已经解释的很明白了，FAQ里也有写，是因为iptables-nft的原因不再支持，想使用可以去用21版本

，感谢，，同时感谢2楼的大佬，原来openwrt.ai  使用Iptables替换Firewall4(Nftables) 的选项是为了兼容，
等学会配置openclash，试试imm的系统，

这里就关闭这个issus了，

closed 2024-08-17 by 142857X1