immortalwrt/luci #438: luci-app-accesscontrol: Internet Access Schedule Control NOT RUNNING(上网时间控制不运行)

CharlesLiaoO, 2024-08-18:
Maintainer: I cannot find. BTW: I cannot find this package in OpenWrt package source, but I can find this package in ImmortalWrt.
Environment: 
- Device: Xiaomi Router 4C
- Version: LuCI openwrt-23.05 branch (git-24.092.44091-d24c2e3) / ImmortalWrt 23.05.3 (r27917-81a1f98d5b)

Description:
Internet Access Schedule Control NOT RUNNING (I've tried reboot)
[image]

Thanks!

1715173329, 2024-08-18:
It's just broken and should be removed from source tree.

CharlesLiaoO, 2024-08-18:
Eh. Is there any alternatives (GUI preferred)?

xlighting2017, 2024-08-18:
> Eh. Is there any alternatives (GUI preferred)?

network -- firewall - rules -- add

closed 2024-08-29 by 1715173329