immortalwrt/luci #455: luci ui显示内存占用和命令行中htop/free不一致

GeekSquirrel, 2024-10-23:
Luci显示占用70%
[image]

htop和free都是占用50%左右

[image]

[image]

今天刚从openwrt迁过来，不太熟悉immortalwrt。这种情况正常吗？

1715173329, 2024-10-23:
显示内容更具体而已。openwrt 自 21.02 以来就用这种显示方式。

closed 2024-10-23 by 1715173329

GeekSquirrel, 2024-10-23:
> 刷新间隔时间不一样吧

不是，我刚才用cat /proc/meminfo看了下，也显示70%，Luci貌似就是引用的这个数据。

现在这个数据稳定地比htop的数据大10%-20%，之前在openwrt没发现有这个现象。

1715173329, 2024-10-23:
减去 buffer 和 cached 就一样了。

GeekSquirrel, 2024-10-23:
> 显示内容更具体而已。openwrt 自 21.02 以来就用这种显示方式。

都是内存占用百分比，数值怎么会稳定相差10%-20%，我之前用的openwrt23.05也没发现有这个问题啊。