immortalwrt/luci #457: x86平台 硬盘休眠luci-app-hd-idle插件报错：TypeError

lysgwl, 2024-10-24:
### Describe the bug

硬盘休眠插件出现以下错误：
TypeError
Cannot read properties of null (reading 'trim')

固件中编译没有包含其他三方插件，都引用了代码库的插件。以下包含了编译的配置文件。

[image]
[image]
[1.txt](https://github.com/user-attachments/files/17503617/1.txt)

### ImmortalWrt version

23.05.04

### ImmortalWrt release

r28061

### ImmortalWrt target/subtarget

399f9a1db3 

### Device

pve  QEMU Standard PC

### Image kind

Self-built image

### Terms

- [X] I am reporting an issue for ImmortalWrt, not an unsupported fork.

1715173329, 2024-10-25:
2874d8efef1aa8b7d6c62d5d5579e6ce4770ffa9

lysgwl, 2024-10-25:
> [2874d8e](https://github.com/immortalwrt/luci/commit/2874d8efef1aa8b7d6c62d5d5579e6ce4770ffa9)

好的，谢谢，有时间测试下。
还有一个问题，能否收录luci-app-filetransfer插件，全网没找见合适版本，翻译都为英文。另外luci-app-accesscontrol插件为什么现在没有了呢？

lysgwl, 2024-10-26:
> [2874d8e](https://github.com/immortalwrt/luci/commit/2874d8efef1aa8b7d6c62d5d5579e6ce4770ffa9)

大佬，问题好像还是没解决，昨晚上新编译的固件，仍然出现这个错误。采用的是云编译。
[image]

1715173329, 2024-10-26:
用隐身模式访问试试

lysgwl, 2024-10-26:
> 用隐身模式访问试试

硬盘休眠插件错误，跟这个有关系？不过关了FullCone NAT功能，启动不再出现：
xt_FULLCONENAT: warning: failed to register a conntrack notifier. Disable active GC for mappings问题了。

硬盘休眠插件错误问题，该如何解决呢？这个插件刚好在用。

lysgwl, 2024-10-27:
> 用隐身模式访问试试

测试了https://firmware-selector.immortalwrt.org/预装的硬盘休眠插件luci-app-hd-idle通用也不能使用。

closed 2024-11-07 by lysgwl