immortalwrt/luci #469: 更新源码编译时这三个插件出错 luci-app-socat、luci-app-openvpn-server、luci-app-ipsec-vpnd

mitsukileung, 2024-11-16:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Actual behaviour

(422/545) Installing strongswan-minimal (5.9.14-r4)
(423/545) Installing strongswan-mod-kernel-libipsec (5.9.14-r4)
(424/545) Installing strongswan-mod-xauth-generic (5.9.14-r4)
(425/545) Installing luci-app-ipsec-vpnd (24.321.31602~ed8665d)
ERROR: luci-app-ipsec-vpnd-24.321.31602~ed8665d: trying to overwrite etc/init.d/ipsec owned by strongswan-ipsec-5.9.14-r4.
(426/545) Installing lucky (2.13.4-r1)
(427/545) Installing luci-app-lucky (2.1.1-r1)
(428/545) Installing minidlna (1.3.3-r1)
(429/545) Installing luci-app-minidlna (24.321.31602~ed8665d)
(430/545) Installing mosdns (5.3.3-r2)
(431/545) Installing v2dat (2022.12.15~47b8ee51-r1)
(432/545) Installing v2ray-geoip (202411140052-r1)
(433/545) Installing v2ray-geosite (20241112092643-r1)
(434/545) Installing luci-app-mosdns (1.6.6-r1)
(435/545) Installing n2n (3.0-r5)
(436/545) Installing luci-app-n2n (3.0.0-r1)
(437/545) Installing natmap (20240813-r2)
(438/545) Installing luci-app-natmap (24.321.31602~ed8665d)
(439/545) Installing npc (0.26.19-r1)
(440/545) Installing luci-app-nps (1.1-r4)
(441/545) Installing ruby (3.3.5-r1)
(442/545) Installing ruby-digest (3.3.5-r1)
(443/545) Installing ruby-enc (3.3.5-r1)
(444/545) Installing ruby-pstore (3.3.5-r1)
(445/545) Installing ruby-bigdecimal (3.3.5-r1)
(446/545) Installing ruby-date (3.3.5-r1)
(447/545) Installing ruby-stringio (3.3.5-r1)
(448/545) Installing ruby-psych (3.3.5-r1)
(449/545) Installing ruby-yaml (3.3.5-r1)
(450/545) Installing unzip (6.0-r8)
(451/545) Installing luci-app-openclash (0.46.050)
(452/545) Installing luci-app-openvpn (24.321.31602~ed8665d)
(453/545) Installing openvpn-easy-rsa (3.1.3-r1)
(454/545) Installing openvpn-openssl (2.6.12-r1)
(455/545) Installing luci-app-openvpn-server (3.0-r0)
ERROR: luci-app-openvpn-server-3.0-r0: trying to overwrite etc/config/openvpn owned by openvpn-openssl-2.6.12-r1.
ERROR: luci-app-openvpn-server-3.0-r0: trying to overwrite etc/easy-rsa/vars owned by openvpn-easy-rsa-3.1.3-r1.
(456/545) Installing microsocks (1.0.4-r1)
(457/545) Installing shadowsocks-rust-sslocal (1.21.0-r1)
(458/545) Installing shadowsocks-rust-ssserver (1.21.0-r1)
(459/545) Installing simple-obfs-client (0.0.5-r1)
(460/545) Installing tcping (0.3-r2)
(461/545) Installing trojan-plus (10.0.3-r2)
(462/545) Installing xray-core (24.11.11-r1)
(463/545) Installing xray-plugin (1.8.24-r1)
(464/545) Installing luci-app-passwall (4.78.4)
(465/545) Installing smartdns (46-r1)
(466/545) Installing luci-app-smartdns (1.2024.45-r1)
(467/545) Installing socat (1.8.0.0-r2)
(468/545) Installing luci-app-socat (24.321.31602~ed8665d)
ERROR: luci-app-socat-24.321.31602~ed8665d: trying to overwrite etc/config/socat owned by socat-1.8.0.0-r2.
ERROR: luci-app-socat-24.321.31602~ed8665d: trying to overwrite etc/init.d/socat owned by socat-1.8.0.0-r2.
(469/545) Installing softethervpn5-libs (5.02.5184-r1)
(470/545) Installing softethervpn5-bridge (5.02.5184-r1)
(471/545) Installing softethervpn5-client (5.02.5184-r1)
(472/545) Installing softethervpn5-server (5.02.5184-r1)
(473/545) Installing luci-app-softethervpn (24.321.31602~ed8665d)
(474/545) Installing tailscale (1.76.6-r1)
(475/545) Installing luci-app-tailscale (1.2.3-r1)
(476/545) Installing ttyd (1.7.3-r1)
(477/545) Installing luci-app-ttyd (24.321.31602~ed8665d)
(478/545) Installing miniupnpd-nftables (2.3.7-r1)
(479/545) Installing luci-app-upnp (24.321.31602~ed8665d)
(480/545) Installing p910nd (0.97-r14)
(481/545) Installing luci-app-usb-printer (1.0-r2)
(482/545) Installing vlmcsd (2020.03.30~e5990804-r1)
(483/545) Installing luci-app-vlmcsd (1.0-r5)
(484/545) Installing vsftpd (3.0.5-r2)
(485/545) Installing luci-app-vsftpd (24.321.31602~ed8665d)
(486/545) Installing openssh-client (9.9_p1-r1)
(487/545) Installing openssh-keygen (9.9_p1-r1)
(488/545) Installing smartmontools (7.4-r1)
(489/545) Installing smartmontools-drivedb (7.4-r1)
(490/545) Installing luci-app-wechatpush (3.5.6-r1)
(491/545) Installing luci-app-wol (24.321.31602~ed8665d)
(492/545) Installing zerotier (1.14.1-r1)
(493/545) Installing luci-app-zerotier (24.321.31602~ed8665d)
(494/545) Installing luci-i18n-adblock-fast-zh-cn (24.321.31602~ed8665d)
(495/545) Installing luci-i18n-airplay2-zh-cn (24.321.31602~ed8665d)
(496/545) Installing luci-i18n-argon-config-zh-cn (24.321.31602~ed8665d)
(497/545) Installing luci-i18n-arpbind-zh-cn (24.321.31602~ed8665d)
(498/545) Installing luci-i18n-autoreboot-zh-cn (24.321.31602~ed8665d)
(499/545) Installing luci-i18n-cloudflared-zh-cn (24.321.31602~ed8665d)
(500/545) Installing luci-i18n-ddns-zh-cn (24.321.31602~ed8665d)
(501/545) Installing luci-i18n-filebrowser-go-zh-cn (24.321.31602~ed8665d)
(502/545) Installing luci-i18n-firewall-zh-cn (24.321.31602~ed8665d)
(503/545) Installing luci-i18n-frpc-zh-cn (24.321.31602~ed8665d)
(504/545) Installing luci-i18n-frps-zh-cn (24.321.31602~ed8665d)
(505/545) Installing luci-i18n-homeproxy-zh-cn (24.303.55221~8ed2f42)
(506/545) Installing luci-i18n-ipsec-vpnd-zh-cn (24.321.31602~ed8665d)
(507/545) Installing luci-i18n-lucky-zh-cn (23.185.13945~79dba51)
(508/545) Installing luci-i18n-minidlna-zh-cn (24.321.31602~ed8665d)
(509/545) Installing luci-i18n-mosdns-zh-cn (24.246.09619~49e5a0d)
(510/545) Installing luci-i18n-n2n-zh-cn (24.321.31602~ed8665d)
(511/545) Installing luci-i18n-natmap-zh-cn (24.321.31602~ed8665d)
(512/545) Installing luci-i18n-nps-zh-cn (24.321.31602~ed8665d)
(513/545) Installing luci-i18n-openvpn-server-zh-cn (24.321.31602~ed8665d)
(514/545) Installing luci-i18n-openvpn-zh-cn (24.321.31602~ed8665d)
(515/545) Installing luci-i18n-package-manager-zh-cn (24.321.31602~ed8665d)
(516/545) Installing luci-i18n-passwall-zh-cn (24.321.31602~ed8665d)
(517/545) Installing luci-i18n-samba4-zh-cn (24.321.31602~ed8665d)
(518/545) Installing luci-i18n-smartdns-zh-cn (24.321.31602~ed8665d)
(519/545) Installing luci-i18n-socat-zh-cn (24.321.31602~ed8665d)
(520/545) Installing luci-i18n-softethervpn-zh-cn (24.321.31602~ed8665d)
(521/545) Installing luci-i18n-tailscale-zh-cn (24.290.39090~ae77b19)
(522/545) Installing luci-i18n-ttyd-zh-cn (24.321.31602~ed8665d)
(523/545) Installing luci-i18n-upnp-zh-cn (24.321.31602~ed8665d)
(524/545) Installing luci-i18n-usb-printer-zh-cn (24.321.31602~ed8665d)
(525/545) Installing luci-i18n-vlmcsd-zh-cn (24.321.31602~ed8665d)
(526/545) Installing luci-i18n-vsftpd-zh-cn (24.321.31602~ed8665d)
(527/545) Installing luci-i18n-wechatpush-zh-cn (24.321.31602~ed8665d)
(528/545) Installing luci-i18n-wol-zh-cn (24.321.31602~ed8665d)
(529/545) Installing luci-i18n-zerotier-zh-cn (24.321.31602~ed8665d)
(530/545) Installing luci-lib-uqr (24.321.31602~ed8665d)
(531/545) Installing wireguard-tools (1.0.20210914-r4)
(532/545) Installing luci-proto-wireguard (24.321.31602~ed8665d)
(533/545) Installing map (7)
(534/545) Installing mkf2fs (1.16.0-r3)
(535/545) Installing mtd (26)
(536/545) Installing odhcp6c (2023.05.12~bcd28363-r20)
(537/545) Installing odhcpd-ipv6only (2024.05.08~a2988231-r2)
(538/545) Installing partx-utils (2.40.2-r1)
(539/545) Installing shellsync (0.2-r2)
(540/545) Installing ppp (2.5.1-r1)
(541/545) Installing ppp-mod-pppoe (2.5.1-r1)
(542/545) Installing procd-ujail (2024.11.13~7330fa55-r1)
(543/545) Installing uci (2023.08.10~5781664d-r1)
(544/545) Installing urandom-seed (3)
(545/545) Installing urngd (2023.11.01~44365eb1-r1)
3 errors; 404 MiB in 545 packages
make[2]: *** [package/Makefile:99: package/install] Error 3

### Expected behaviour

编译时出错

### Steps to reproduce

编译时出错

### Additional Information

```shell
X86架构
master分支
```

Firsgith, 2024-12-22:
luci-app-ipsec-vpnd和server二选一吧，插件间冲突，没见解决。socat和那个luciappsocat也是。

jessons, 2025-02-07:
(219/277) Installing openssl-util (3.0.15-r1)
(220/277) Installing openvpn-easy-rsa (3.2.1-r1)
(221/277) Installing openvpn-openssl (2.6.13-r1)
(222/277) Installing luci-app-openvpn-server (3.0-r0)
ERROR: luci-app-openvpn-server-3.0-r0: trying to overwrite etc/config/openvpn owned by openvpn-openssl-2.6.13-r1.
ERROR: luci-app-openvpn-server-3.0-r0: trying to overwrite etc/easy-rsa/vars owned by openvpn-easy-rsa-3.2.1-r1.
(223/277) Installing luci-compat (25.033.43316~cde8dde)
我也有这个问题，我只选择了一个openvpn，没选择其他的VPN应用，

helllkz, 2025-03-26:
我也发现了，openvpn-server会导致编译报错
https://github.com/immortalwrt/immortalwrt/discussions/1768

closed 2025-03-27 by 1715173329