immortalwrt/luci #472: luci-app-alist 无法重复启动

ximiTech, 2024-11-27:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Actual behaviour

在luci界面中配置使能后，进网页配置，然后再在界面中取消启用，保存应用
之后，如果再想使能启用，就起不来了，
第一次报错 not enable search，就又试着使能了 search
然后，没报错了，但是还是不能重复启动

### Expected behaviour

关闭后能重复启动

### Steps to reproduce

在luci界面中配置使能后，进网页配置，然后再在界面中取消启用，保存应用
之后，如果再想使能启用，就起不来了，
第一次报错 not enable search，就又试着使能了 search
然后，没报错了，但是还是不能重复启动

### Additional Information

```shell
OpenWrt SNAPSHOT
BUILD_ID="r28181-faf4b3e0f7"
```

### What browsers do you see the problem on?

Chrome

ximiTech, 2024-11-29:
又测试了一下，发现是由于设置了监听地址为一个非默认的以 .0 结尾的地址时 就会出现启动不了的情况

1715173329, 2024-12-03:
alist 配置，系统日志

closed 2025-02-18 by 1715173329