immortalwrt/luci #474: 官方X86固件，安装passwall后luci-app-adblock不能去广告

sclsyin, 2024-12-03:
### Describe the bug

官方X86固件，安装passwall后luci-app-adblock不能去广告

### ImmortalWrt version

r28061-399f9a1db3

### ImmortalWrt release

23.05.4

### ImmortalWrt target/subtarget

x86/64

### Device

QEMU Standard PC (Q35 + ICH9, 2009)

### Image kind

Official downloaded image

### Steps to reproduce

安装passwall后adblock广告拦截不能去广告

### Actual behaviour

adblock广告拦截不能去广告

### Expected behaviour

能去广告

### Additional info

[image]
[image]

### Terms

- [X] I am reporting an issue for ImmortalWrt, not an unsupported fork.

1715173329, 2024-12-03:
说明这两个插件有冲突，只能开一个。

closed 2024-12-03 by 1715173329

sclsyin, 2024-12-03:
> 说明这两个插件有冲突，只能开一个。

刚刚全新导入，只安装Adblock也不能过滤。。。
[image]
[image]
[image]

1715173329, 2024-12-03:
你指望 DNS 去广告能有什么效果，要有效果用浏览器插件和客户端。

sclsyin, 2024-12-03:
额，之前旧版本的效果还是不错的呀。