immortalwrt/luci #481: luci-app-socat构建固件错误，在线安装luci-app-socat也出错

vison-v, 2024-12-12:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### screenshots or captures

[image]

### Actual behaviour

在线构建固件只添加luci-app-socat一个软件包时报错：
Generate local signing keys...
WARNING: can't open config file: /mnt/disk/master/staging_dir/host/etc/ssl/openssl.cnf
WARNING: can't open config file: /mnt/disk/master/staging_dir/host/etc/ssl/openssl.cnf
read EC key
writing EC key
WARNING: opening packages/packages.adb: No such file or directory
Package list missing or not up-to-date, generating it.

Building package index...
ERROR: luci-app-socat-24.129.54850~7782759: trying to overwrite etc/config/socat owned by socat-1.8.0.0-r2.
ERROR: luci-app-socat-24.129.54850~7782759: trying to overwrite etc/init.d/socat owned by socat-1.8.0.0-r2.
make[2]: *** [Makefile:225: package_install] Error 1
make[1]: *** [Makefile:166: _call_manifest] Error 2
make: *** [Makefile:340: manifest] Error 2

### Expected behaviour

固件不能构建

### Steps to reproduce

在官方在线构建固件或actions编译都是同样的错误

### Additional Information

```shell
NAME="ImmortalWrt"
VERSION="SNAPSHOT"
ID="immortalwrt"
ID_LIKE="lede openwrt"
PRETTY_NAME="ImmortalWrt SNAPSHOT"
VERSION_ID="snapshot"
HOME_URL="https://immortalwrt.org/"
BUG_URL="https://github.com/immortalwrt/immortalwrt/issues"
SUPPORT_URL="https://github.com/immortalwrt/immortalwrt/discussions"
BUILD_ID="r0-4df64a9"
OPENWRT_BOARD="x86/64"
OPENWRT_ARCH="x86_64"
OPENWRT_TAINTS="no-all"
OPENWRT_DEVICE_MANUFACTURER="ImmortalWrt"
OPENWRT_DEVICE_MANUFACTURER_URL="https://immortalwrt.org/"
OPENWRT_DEVICE_PRODUCT="Generic"
OPENWRT_DEVICE_REVISION="v0"
OPENWRT_RELEASE="ImmortalWrt SNAPSHOT r0-4df64a9"
OPENWRT_BUILD_DATE="1733897848"
```

### What browsers do you see the problem on?

Microsoft Edge

lambdaplus, 2024-12-12:
是的，我昨天也遇到了，24.10.0-rc1分支，编译时添加socat就会报错，去掉就编译成功。使用的imagebuilder

Firsgith, 2024-12-22:
luci-app-socat会和它的依赖包socat冲突，官方没解决。试试按好早之前的pr试试能解决不。https://github.com/immortalwrt/luci/pull/205

myfalcom, 2024-12-22:
24.10 版本即使在线安装完成后，配置文件也有变动。应该是配置兼容问题。
建议更新源代码适配。

fairok, 2025-02-19:
各位大佬，这个有临时解决办法吗？

1715173329, 2025-03-27:
鉴于此 luci 未得到妥善维护，现已从库中移除。

closed 2025-03-27 by 1715173329

jjm2473, 2025-06-13:
istore有支持fw4的版本，且无文件冲突问题。

myfalcom, 2025-06-13:
和FW4关系不大，主要是写入的配置和老版的不一致了。如果参考新版配置自己写的话，socat完全可以起得来的。

hour-keeper, 2025-06-30:
@jiesou 可以用天灵开发的 luci-app-3cat