immortalwrt/luci #485: [Bug] openwrt-24.10 防火墙-常规设置  无法正常保存

NextAbyss, 2024-12-21:
### Describe the bug

拉取的最新commit 0f7a2d3,防火墙-常规设置无法保存，症状是一直转圈圈，且右上角没有出现未保存的配置提示。
[image]

### ImmortalWrt version

ImmortalWrt 24.10-SNAPSHOT (r28208-0f7a2d3e21)

### ImmortalWrt release

ImmortalWrt 24.10-SNAPSHOT (r28208-0f7a2d3e21)

### ImmortalWrt target/subtarget

x86-64

### Device

PVE

### Image kind

Self-built image

### Steps to reproduce

网络-防火墙-常规设置,前面几项能打勾的地方随便勾一下，路由/NAT 卸载这里也可以随便选一个,点击保存无法正常保存。

### Terms

- [X] I am reporting an issue for ImmortalWrt, not an unsupported fork.

mitsukileung, 2024-12-22:
master分支也有这样的问题

1715173329, 2024-12-22:
fixed in 8a422af6a0e97cdb7b3df1049898731a4b2d153d

closed 2024-12-22 by 1715173329