immortalwrt/luci #490: ACME安装报错

SuitThug3977, 2024-12-26:
### Describe the bug

安装acme，安装acme-common这个包的时候报错：
/etc/rc.common: eval: line 111: /usr/lib/acme/hook: not found
acme: cleaning up

acme界面无法保存dnsapi参数（保存无效一刷新就没有了）

### ImmortalWrt version

r32632-59ac244b6d

### ImmortalWrt release

SNAPSHOT

### ImmortalWrt target/subtarget

x86/64

### Device

Microsoft Corporation Virtual Machine

### Image kind

Self-built image

### Steps to reproduce

软件包里搜索acme安装

### Terms

- [X] I am reporting an issue for ImmortalWrt, not an unsupported fork.

wi11iamzhao, 2024-12-26:
acmesh-dnsapi安装了么

SuitThug3977, 2024-12-27:
装过了啊，acme-common这个包是最基础的包，这个包安装出错别的就不用想了。。lede的没这个问题。