immortalwrt/luci #495: 求助：从某次更新后，passwall一直编译失败，trojan-plus这个一直编译失败

lfsakura0, 2025-01-12:
### Is there an existing issue for this?

- [X] I have searched among all existing issues (including closed issues)

### screenshots or captures

错误日志截图：
[image]
[image]

完整的编译日志文件：
[build-log.txt](https://github.com/user-attachments/files/18388614/build-log.txt)

config文件：
[config.txt](https://github.com/user-attachments/files/18388609/config.txt)

### Actual behaviour

只要勾选passwall编译，一定会是 ERROR: package/feeds/packages/trojan-plus failed to build 这个原因报错，已经重复很多次了，不勾选passwall则可以编译成功

### Expected behaviour

之前是直接编译成功的，某次./scripts/feeds update -a && ./scripts/feeds install -a更新后，选择passwall就会失败

### Steps to reproduce

勾选passwall编译

### Additional Information

```shell
主机名	ImmortalWrt
型号	QEMU Standard PC (Q35 + ICH9, 2009)
架构	Intel(R) Celeron(R) J4125 CPU @ 2.00GHz
目标平台	x86/64
固件版本	ImmortalWrt SNAPSHOT r32925-d3f097a70e / LuCI Master 25.010.28961~6abdc2a
内核版本	6.6.69
immortalwrt分支：master
```

### What browsers do you see the problem on?

Microsoft Edge

yehualiu1979, 2025-01-14:
trojan-plus编辑了半个月，都是和你一样的情况。
最后，决定不要这个了。
编译出来的passwall还是能满足我的要求。

修改config文件
CONFIG_PACKAGE_luci-app-passwall_INCLUDE_Trojan_Plus=n
CONFIG_PACKAGE_trojan-plus=n

或者#剔除Trojan_Plus
echo "CONFIG_PACKAGE_luci-app-passwall_INCLUDE_Trojan_Plus=n" >> ./.config
echo "CONFIG_PACKAGE_trojan-plus=n" >> ./.config

lfsakura0, 2025-01-14:
> trojan-plus编辑了半个月，都是和你一样的情况。 最后，决定不要这个了。 编译出来的passwall还是能满足我的要求。
>
> 修改config文件 CONFIG_PACKAGE_luci-app-passwall_INCLUDE_Trojan_Plus=n CONFIG_PACKAGE_trojan-plus=n
>
> 或者#剔除Trojan_Plus echo [...]
[image]
我用gpt给出的提示是这样的，但trojan-plus-10.0.3好像是编译的时候才下载，make clean之后就无了。我现在用mihomo和openclash，等有解决方法再用pw了。