immortalwrt/luci #505: 请求添加 luci-app-nikki

Lu-Lucifer, 2025-02-10:
原来的luci-app-mihomo改名，请求添加包luci-app-nikki（https://github.com/nikkinikki-org/OpenWrt-nikki）

Firsgith, 2025-06-02:
我尝试过自己添加进去，装进去之后核心状态都是“未在运行”，手动开启也没用

msylgj, 2025-06-02:
@Firsgith https://github.com/nikkinikki-org/OpenWrt-nikki   Read 'README'

jiesou, 2025-07-05:
加入官方源的话，用起来能方便很多
升级固件能直接在线构建
而且 nikki 的 pages.dev 源老是被墙的

Zhang-Siyang, 2025-07-16:
我想帮助添加到feeds中，请问有谁知道该提交什么样的PR吗？

jiesou, 2025-09-05:
> > 我想帮助添加到feeds中，请问有谁知道该提交什么样的PR吗？
> 
> # 对方不是说了么，添加源
> 
> echo "src-git nikki https://github.com/nikkinikki-org/OpenWrt-nikki.git;main" >> "feeds.conf.default"
> # 更新并安装源
> 
> ./scripts/feeds update -a ./scripts/feeds install -a
> # 编译
> 
> make package/luci-app-nikki/compile

加到官方源和第三方源能一样吗

sgpublic, 2025-09-19:
> 加到官方源和第三方源能一样吗

我觉得没啥区别，不都得自己编译，写个脚本就加进去了