immortalwrt/luci #520: Cannot read properties of null (reading 'wifi')

siwee, 2025-03-16:
### Describe the bug

[image]

### ImmortalWrt version

r32824-6a73dae98c9c

### ImmortalWrt release

24.10.0

### ImmortalWrt target/subtarget

x86/64

### Device

VMware, Inc. VMware Virtual Platform

### Image kind

Self-built image

### Diffconfig

```properties
CONFIG_TARGET_x86=y
CONFIG_TARGET_x86_64=y
CONFIG_TARGET_x86_64_DEVICE_generic=y
CONFIG_GNUTLS_ALPN=y
CONFIG_GNUTLS_ANON=y
CONFIG_GNUTLS_DTLS_SRTP=y
CONFIG_GNUTLS_HEARTBEAT=y
CONFIG_GNUTLS_OCSP=y
CONFIG_GNUTLS_PSK=y
CONFIG_HTOP_LMSENSORS=y
CONFIG_LIBCURL_COOKIES=y
CONFIG_LIBCURL_CRYPTO_AUTH=y
CONFIG_LIBCURL_FILE=y
CONFIG_LIBCURL_FTP=y
CONFIG_LIBCURL_HTTP=y
CONFIG_LIBCURL_NGHTTP2=y
CONFIG_LIBCURL_OPENSSL=y
CONFIG_LIBCURL_PROXY=y
CONFIG_LIBCURL_TFTP=y
CONFIG_LIBCURL_THREADED_RESOLVER=y
CONFIG_LIBCURL_TLS_SRP=y
CONFIG_LIBCURL_UNIX_SOCKETS=y
CONFIG_PACKAGE_attr=y
CONFIG_PACKAGE_avahi-dbus-daemon=y
# CONFIG_PACKAGE_bnx2-firmware is not set
CONFIG_PACKAGE_btop=y
CONFIG_PACKAGE_curl=y
CONFIG_PACKAGE_dbus=y
CONFIG_PACKAGE_ddns-scripts=y
CONFIG_PACKAGE_ddns-scripts-cloudflare=y
CONFIG_PACKAGE_ddns-scripts-services=y
CONFIG_PACKAGE_ddns-scripts_aliyun=y
CONFIG_PACKAGE_dnsmasq_full_broken_rtc=y
# CONFIG_PACKAGE_dnsmasq_full_dhcp is not set
CONFIG_PACKAGE_dnsmasq_full_ipset=y
CONFIG_PACKAGE_etherwake=y
CONFIG_PACKAGE_htop=y
CONFIG_PACKAGE_ip-tiny=y
CONFIG_PACKAGE_iperf3=y
CONFIG_PACKAGE_ipset=y
# CONFIG_PACKAGE_kmod-8139cp is not set
# CONFIG_PACKAGE_kmod-8139too is not set
# CONFIG_PACKAGE_kmod-amazon-ena is not set
# CONFIG_PACKAGE_kmod-amd-xgbe is not set
# CONFIG_PACKAGE_kmod-bnx2 is not set
# CONFIG_PACKAGE_kmod-drm-buddy is not set
# CONFIG_PACKAGE_kmod-drm-display-helper is not set
# CONFIG_PACKAGE_kmod-drm-i915 is not set
# CONFIG_PACKAGE_kmod-drm-kms-helper is not set
# CONFIG_PACKAGE_kmod-drm-ttm is not set
# CONFIG_PACKAGE_kmod-dwmac-intel is not set
# CONFIG_PACKAGE_kmod-e1000e is not set
# CONFIG_PACKAGE_kmod-fb is not set
# CONFIG_PACKAGE_kmod-fb-cfb-copyarea is not set
# CONFIG_PACKAGE_kmod-fb-cfb-fillrect is not set
# CONFIG_PACKAGE_kmod-fb-cfb-imgblt is not set
# CONFIG_PACKAGE_kmod-fb-sys-fops is not set
# CONFIG_PACKAGE_kmod-fb-sys-ram is not set
# CONFIG_PACKAGE_kmod-forcedeth is not set
# CONFIG_PACKAGE_kmod-hid is not set
# CONFIG_PACKAGE_kmod-hid-generic is not set
# CONFIG_PACKAGE_kmod-i40e is not set
# CONFIG_PACKAGE_kmod-igbvf is not set
# CONFIG_PACKAGE_kmod-igc is not set
CONFIG_PACKAGE_kmod-inet-diag=y
# CONFIG_PACKAGE_kmod-input-evdev is not set
CONFIG_PACKAGE_kmod-ipt-core=y
CONFIG_PACKAGE_kmod-ipt-ipset=y
# CONFIG_PACKAGE_kmod-ixgbe is not set
# CONFIG_PACKAGE_kmod-ixgbevf is not set
# CONFIG_PACKAGE_kmod-libphy is not set
# CONFIG_PACKAGE_kmod-mii is not set
# CONFIG_PACKAGE_kmod-net-selftests is not set
CONFIG_PACKAGE_kmod-netlink-diag=y
CONFIG_PACKAGE_kmod-nf-ipt=y
CONFIG_PACKAGE_kmod-nf-tproxy=y
CONFIG_PACKAGE_kmod-nft-tproxy=y
# CONFIG_PACKAGE_kmod-pcnet32 is not set
# CONFIG_PACKAGE_kmod-pcs-xpcs is not set
# CONFIG_PACKAGE_kmod-phy-ax88796b is not set
# CONFIG_PACKAGE_kmod-phylink is not set
# CONFIG_PACKAGE_kmod-r8101 is not set
# CONFIG_PACKAGE_kmod-r8125 is not set
# CONFIG_PACKAGE_kmod-r8126 is not set
# CONFIG_PACKAGE_kmod-r8168 is not set
# CONFIG_PACKAGE_kmod-stmmac-core is not set
# CONFIG_PACKAGE_kmod-tg3 is not set
# CONFIG_PACKAGE_kmod-tulip is not set
CONFIG_PACKAGE_kmod-tun=y
# CONFIG_PACKAGE_kmod-usb-hid is not set
# CONFIG_PACKAGE_kmod-usb-net is not set
# CONFIG_PACKAGE_kmod-usb-net-asix is not set
# CONFIG_PACKAGE_kmod-usb-net-asix-ax88179 is not set
# CONFIG_PACKAGE_kmod-usb-net-rtl8150 is not set
# CONFIG_PACKAGE_kmod-usb-net-rtl8152-vendor is not set
# CONFIG_PACKAGE_kmod-vmxnet3 is not set
CONFIG_PACKAGE_libatomic=y
CONFIG_PACKAGE_libattr=y
CONFIG_PACKAGE_libavahi-client=y
CONFIG_PACKAGE_libavahi-dbus-support=y
CONFIG_PACKAGE_libcap=y
CONFIG_PACKAGE_libcap-ng=y
CONFIG_PACKAGE_libcurl=y
CONFIG_PACKAGE_libdaemon=y
CONFIG_PACKAGE_libdbus=y
CONFIG_PACKAGE_libexpat=y
CONFIG_PACKAGE_libgnutls=y
CONFIG_PACKAGE_libiperf3=y
CONFIG_PACKAGE_libipset=y
CONFIG_PACKAGE_libminiupnpc=y
CONFIG_PACKAGE_libnatpmp=y
CONFIG_PACKAGE_libnghttp2=y
CONFIG_PACKAGE_libopenssl-conf=y
CONFIG_PACKAGE_libpcre2=y
CONFIG_PACKAGE_libpopt=y
CONFIG_PACKAGE_libreadline=y
CONFIG_PACKAGE_libstdcpp=y
CONFIG_PACKAGE_libtasn1=y
CONFIG_PACKAGE_libtirpc=y
CONFIG_PACKAGE_liburing=y
CONFIG_PACKAGE_libuv=y
CONFIG_PACKAGE_libwebsockets-full=y
CONFIG_PACKAGE_luci-app-ddns=y
# CONFIG_PACKAGE_luci-app-diskman_INCLUDE_btrfs_progs is not set
# CONFIG_PACKAGE_luci-app-diskman_INCLUDE_lsblk is not set
# CONFIG_PACKAGE_luci-app-rclone_INCLUDE_rclone-ng is not set
# CONFIG_PACKAGE_luci-app-rclone_INCLUDE_rclone-webui is not set
CONFIG_PACKAGE_luci-app-samba4=y
CONFIG_PACKAGE_luci-app-ttyd=y
CONFIG_PACKAGE_luci-app-upnp=y
CONFIG_PACKAGE_luci-app-vlmcsd=y
CONFIG_PACKAGE_luci-app-wol=y
CONFIG_PACKAGE_luci-app-zerotier=y
CONFIG_PACKAGE_luci-i18n-ddns-zh-cn=y
CONFIG_PACKAGE_luci-i18n-samba4-zh-cn=y
CONFIG_PACKAGE_luci-i18n-ttyd-zh-cn=y
CONFIG_PACKAGE_luci-i18n-upnp-zh-cn=y
CONFIG_PACKAGE_luci-i18n-vlmcsd-zh-cn=y
CONFIG_PACKAGE_luci-i18n-wol-zh-cn=y
CONFIG_PACKAGE_luci-i18n-zerotier-zh-cn=y
CONFIG_PACKAGE_luci-theme-argon=y
CONFIG_PACKAGE_miniupnpd-nftables=y
CONFIG_PACKAGE_odhcpd=y
# CONFIG_PACKAGE_odhcpd-ipv6only is not set
CONFIG_PACKAGE_odhcpd_full_ext_cer_id=0
CONFIG_PACKAGE_openssl-util=y
CONFIG_PACKAGE_samba4-libs=y
CONFIG_PACKAGE_samba4-server=y
CONFIG_PACKAGE_sing-box=y
CONFIG_PACKAGE_ttyd=y
CONFIG_PACKAGE_vlmcsd=y
CONFIG_PACKAGE_wget-ssl=y
CONFIG_PACKAGE_wsdd2=y
CONFIG_PACKAGE_zerotier=y
CONFIG_PACKAGE_zlib=y
CONFIG_PCRE2_JIT_ENABLED=y
CONFIG_SAMBA4_SERVER_AVAHI=y
CONFIG_SAMBA4_SERVER_NETBIOS=y
CONFIG_SAMBA4_SERVER_VFS=y
CONFIG_SAMBA4_SERVER_WSDD2=y
CONFIG_SING_BOX_BUILD_CLASH_API=y
CONFIG_SING_BOX_BUILD_DHCP=y
CONFIG_SING_BOX_BUILD_ECH=y
CONFIG_SING_BOX_BUILD_GRPC=y
CONFIG_SING_BOX_BUILD_GVISOR=y
CONFIG_SING_BOX_BUILD_QUIC=y
CONFIG_SING_BOX_BUILD_REALITY_SERVER=y
CONFIG_SING_BOX_BUILD_UTLS=y
CONFIG_VMDK_IMAGES=y
# CONFIG_SING_BOX_BUILD_ACME is not set
# CONFIG_SING_BOX_BUILD_WIREGUARD is not set
```

### Terms

- [x] I am reporting an issue for ImmortalWrt, not an unsupported fork.

siwee, 2025-03-17:
这个错误提示并不是每次都会出现

1715173329, 2025-03-17:
只会在登录或者会话超时的时候出现，刷新一下就没了。感觉像上游的 bug 但那边又没人开 issue。

siwee, 2025-03-24:
> 只会在登录或者会话超时的时候出现，刷新一下就没了。感觉像上游的 bug 但那边又没人开 issue。

是的，我用OpenWrt官方固件复现了这个问题并开了一个issue https://github.com/openwrt/luci/issues/7698

SqiChiu, 2025-05-15:
luci那边修改了代码，但是问题没解决，不知道为什么，lean、kwrt的固件都没有这个问题