immortalwrt/luci #525: <luci-app-filebrowser>: <包内容有误>

superpassby, 2025-03-29:
### Is there an existing issue for this?

- [x] I have searched among all existing issues (including closed issues)

### screenshots or captures

### Actual behaviour

master 及 openwrt-24.10 这两个分支中的 luci/applications/luci-app-filebrowser：
LUCI_TITLE:=LuCI File Browser module
该包不能配合 packages/utils/filebrowser (PKG_SOURCE:https://github.com/filebrowser/filebrowser) 使用。

而 openwrt-23.05 及之前的分支中的 luci/applications/luci-app-filebrowser
LUCI_TITLE:=LuCI app for FileBrowser
这个才是 packages/utils/filebrowser 的 luci 程序。

### Expected behaviour

请调整 master 及 openwrt-24.10 这两个分支中的 luci/applications/luci-app-filebrowser 内容。

注：
luci/applications/luci-app-filebrowser：
LUCI_TITLE:=LuCI File Browser module

是官方源在 24.10 版本新加入的包，与 immortalwrt 仓库现有的 luci-app-filebrowser 包重名，建议对2者的包名进行区分；考虑到官方源的 luci-app-filebrowser 功能非常有限，且与 luci-app-filemanager 包功能重复，建议只保留本仓库的luci-app-filebrowser（LuCI app for FileBrowser）。

### Steps to reproduce

略，与本次提交无关。

### Additional Information

```shell
略，与本次提交无关。
```

### Relevant log output

```shell

```

1715173329, 2025-03-29:
现在叫 luci-app-filebrowser-go。

closed 2025-03-29 by 1715173329

caixin1218, 2025-03-31:
> 现在叫 luci-app-filebrowser-go。

是的，但是master和24.10 的分支根本无法使用，虽然可以正常启动，但是打开web界面显示404 page not found

1715173329, 2025-04-01:
这问题两个月前就修了

caixin1218, 2025-04-02:
> 这问题两个月前就修了
系统日志：
Wed Apr  2 17:34:33 2025 daemon.err filebrowser[11672]: 2025/04/02 17:34:33 Warning: filebrowser.db can't be found. Initialing in /etc/filebrowser/database.db
Wed Apr  2 17:34:33 2025 daemon.err filebrowser[11672]: 2025/04/02 17:34:33 Using database: /etc/filebrowser/database.db
Wed Apr  2 17:34:33 2025 daemon.err filebrowser[11672]: 2025/04/02 17:34:33 No config file used
Wed Apr  2 17:34:33 2025 daemon.info filebrowser[11672]: 2025/04/02 17:34:33 Listening on [::]:8989
Wed Apr  2 17:34:47 2025 daemon.info filebrowser[11672]: 2025/04/02 17:34:47 /: 404 192.168.18.7 open public/index.html: file does not exist
Wed Apr  2 17:34:47 2025 daemon.info filebrowser[11672]: 2025/04/02 17:34:47 /favicon.ico: 404 192.168.18.7 open public/index.html: file does not exist
Wed Apr  2 17:34:55 2025 daemon.info filebrowser[11672]: 2025/04/02 17:34:55 Caught signal terminated: shutting down.
Wed Apr  2 17:34:55 2025 daemon.info filebrowser[11672]: 2025/04/02 17:34:55 accept tcp [::]:8989: use of closed network connection
Wed Apr  2 17:34:55 2025 daemon.err filebrowser[11881]: 2025/04/02 17:34:55 Using database: /etc/filebrowser/database.db
Wed Apr  2 17:34:55 2025 daemon.err filebrowser[11881]: 2025/04/02 17:34:55 No config file used
Wed Apr  2 17:34:55 2025 daemon.info filebrowser[11881]: 2025/04/02 17:34:55 Listening on [::]:8989
Wed Apr  2 17:35:01 2025 daemon.info filebrowser[11881]: 2025/04/02 17:35:01 /: 404 192.168.18.7 open public/index.html: file does not exist

架构	AMD A4 Micro-6400T APU + AMD Radeon R3 Graphics x 4C 4T (997.999MHz)
目标平台	x86/64
固件版本	ImmortalWrt 24.10-SNAPSHOT r33004-bca880a0e8 / LuCI openwrt-24.10 branch 25.087.19551~6da215f
内核版本	6.6.83
本地时间	2025-04-02 17:37:00

caixin1218, 2025-04-06:
> 这问题两个月前就修了

大佬请问我编译的openwrt-24.10 luci-app-filebrowser-go为什么会出现404 web not found ，编译过程都是正常的，也不缺少依赖，实在是找不到原因和解决办法，请指教。