immortalwrt/luci #528: passwall 不能正常安装使用

404area, 2025-04-05:
### Describe the bug

缺少依赖

### ImmortalWrt version

23.0.x

### ImmortalWrt release

r28061-399f9a1db3

### ImmortalWrt target/subtarget

armsr/armv8

### Device

passwall

### Image kind

Official downloaded image

### Steps to reproduce

安装passwall 总是提示依赖的软包 shadowsocks-rust-sslocal在所在的仓库都未提供，所有的23.x版本都是，包括docker镜像和虚拟磁盘镜像，只要是新安装就提示没有这个依赖

### Diffconfig

```text

```

### Terms

- [x] I am reporting an issue for ImmortalWrt, not an unsupported fork.

404area, 2025-04-05:
> 强制安装 passwall 然后插件选singbox 基本上就够了。您的设备是什么？

而且很可能几乎是这个大版本都有这个问题，包括docker.磁盘映像，其他版本暂时不确定，不知道是不是passwall 版本升级了，2月份还没这个问题

libriunk, 2025-04-05:
> > 强制安装 passwall 然后插件选singbox 基本上就够了。您的设备是什么？
> 
> 而且很可能几乎是这个大版本都有这个问题，包括docker.磁盘映像，其他版本暂时不确定，不知道是不是passwall 版本升级了，2月份还没这个问题

**不建议在** immo/op的2305及之后的版本安装docker

404area, 2025-04-05:
> > > 强制安装 passwall 然后插件选singbox 基本上就够了。您的设备是什么？
> > 
> > 
> > 而且很可能几乎是这个大版本都有这个问题，包括docker.磁盘映像，其他版本暂时不确定，不知道是不是passwall 版本升级了，2月份还没这个问题
> 
> **不建议在** immo/op的2305及之后的版本安装docker

您可能理解错了，我的意思是说用容器或者磁盘映像的方式安装op,然后安装passwall 就会报这个问题，而不是在op 中安装docker

404area, 2025-04-05:
我现在只能尝试使用are you ok提供一件脚本安装，看看是否可以，我猜测大概率是passwall更新了，导致的原因

404area, 2025-04-05:
> 我现在只能尝试使用are you ok提供一件脚本安装，看看是否可以，我猜测大概率是passwall更新了，导致的原因

23.x没有这个依赖的源，24.x是有的

closed 2025-04-08 by 1715173329