immortalwrt/luci #529: 网络——接口，编辑下，DHCP服务器的常规设置菜单逻辑问题

xingmiye, 2025-04-08:
### Describe the bug

网络——接口，编辑下，DHCP服务器的常规设置：
感觉如果勾选了忽略此接口，就是不再此接口提供DHCP服务，那么下方的启动、客户数、租约时间可以留空甚至可以直接隐藏掉

[image]

以前23.05.4，租约时间还可以留空不填的，但是到了24.10，却变成了必填选项。
如果不勾选忽略此接口，使用dhcp服务，确实应该必填。
但是如果不启用dhcp服务，启动、客户数、租约时间三个选项的配置应该是无效的，可以留空或者隐藏的。

### ImmortalWrt version

DISTRIB_REVISION='r32824-6a73dae98c9c'

### ImmortalWrt release

DISTRIB_RELEASE='24.10.0'

### ImmortalWrt target/subtarget

DISTRIB_TARGET='x86/64'

### Device

QEMU Standard PC (i440FX + PIIX, 1996)

### Image kind

Self-built image

### Diffconfig

```text

```

### Terms

- [x] I am reporting an issue for ImmortalWrt, not an unsupported fork.

libriunk, 2025-04-08:
你应该把这个建议发到到op那边

xingmiye, 2025-04-11:
> 你应该把这个建议发到到op那边

好的，发了

xingmiye, 2025-04-11:
> 你应该把这个建议发到到op那边

但是我感觉可能op那边会懒得改，因为op那边主要面对的用户都是把op作为主路由的，DHCP服务都开着的。

libriunk, 2025-04-11:
你用刷op的固件反应这个问题，戈门，你拿个immo的图直接过去，这不closed？