immortalwrt/luci #533: luci-app-openvpn-server 仍然有冲突

Fachep, 2025-04-19:
Package luci-app-openvpn-server wants to install file /builder/build_dir/target-aarch64_cortex-a53_musl/root-mediatek/etc/config/openvpn
	But that file is already provided by package  * openvpn-openssl

asu构建结果：
请求数据：
```json
{
    "url": "https://sysupgrade.kyarucloud.moe",
    "branch": "24.10",
    "revision": "r32824-6a73dae98c9c",
    "efi": null,
    "advanced_mode": "1",
    "request_hash": "0f540065f0a2fb2f046645a85403e8819aa9e0a1e735e1bccc131fd872329240",
    "sha256_unsigned": "",
    "client": "luci/25.044.30833~39a1ba3",
    "packages": {
        "iwinfo": "2024.10.20~b94f066e-r1",
        "luci-mod-rpc": "25.040.36223~e888ad7",
        "kmod-lib-textsearch": "6.6.73-r1",
        "libc": "1.2.5-r4",
        "kmod-macvlan": "6.6.73-r1",
        "luci-i18n-argon-config-zh-cn": "25.040.36223~e888ad7",
        "opkg": "2024.10.16~38eccbb1-r1",
        "ip-full": "6.11.0-r1",
        "libubus-lua": "2025.01.02~afa57cce-r1",
        "kmod-crypto-aead": "6.6.73-r1",
        "kmod-lib-crc-ccitt": "6.6.73-r1",
        "getrandom": "2024.04.26~85f10530-r1",
        "bind-host": "9.20.4-r1",
        "kmod-phy-aquantia": "6.6.73-r1",
        "kmod-asn1-decoder": "6.6.73-r1",
        "libustream-openssl": "2024.07.28~99bd3d2b-r1",
        "kmod-pppoe": "6.6.73-r1",
        "kmod-nf-nathelper": "6.6.73-r1",
        "procd-ujail": "2024.12.22~42d39376-r1",
        "base-files": "1844~6a73dae98c9c",
        "luci-nginx": "25.040.36223~e888ad7",
        "fitblk": "2",
        "ddns-scripts": "2.8.2-r52",
        "luci-i18n-openvpn-server-zh-cn": "25.040.36223~e888ad7",
        "uboot-envtools": "2024.07-r1",
        "autocore": "42",
        "kmod-crypto-acompress": "6.6.73-r1",
        "bridger": "2024.04.22~40b1c5b6",
        "block-mount": "2024.07.14~408c2cc4-r1",
        "libopenssl-legacy": "3.0.15-r1",
        "luci-lib-ipkg": "25.040.36223~e888ad7",
        "wpad-openssl": "2024.09.15~5ace39b0-r2",
        "odhcp6c": "2024.09.25~b6ae9ffa-r1",
        "uci": "2025.01.20~16ff0bad-r1",
        "lua": "5.1.5-r11",
        "dropbear": "2024.86-r1",
        "kmod-hwmon-core": "6.6.73-r1",
        "luci-compat": "25.040.36223~e888ad7",
        "mtd": "26",
        "odhcpd-ipv6only": "2024.05.08~a2988231-r2",
        "i2c-tools": "4.3-r3",
        "kmod-crypto-hw-safexcel": "6.6.73-r1",
        "kmod-crypto-authenc": "6.6.73-r1",
        "kmod-mt7915e": "6.6.73.2025.01.14~8e4f72b6-r1",
        "urandom-seed": "3",
        "luci-i18n-package-manager-zh-cn": "25.040.36223~e888ad7",
        "ppp": "2.5.1-r1",
        "kmod-leds-gpio": "6.6.73-r1",
        "kmod-gpio-button-hotplug": "6.6.73-r5",
        "logd": "2024.04.26~85f10530-r1",
        "ddns-scripts-dnspod-v3": "2.8.2-r64",
        "kmod-crypto-ecb": "6.6.73-r1",
        "luci-app-attendedsysupgrade": "25.044.30833~39a1ba3",
        "luci-i18n-ddns-zh-cn": "25.040.36223~e888ad7",
        "kmod-libphy": "6.6.73-r1",
        "kmod-crypto-des": "6.6.73-r1",
        "kmod-nf-nathelper-extra": "6.6.73-r1",
        "luci-i18n-acme-zh-cn": "25.040.36223~e888ad7",
        "luci-i18n-wol-zh-cn": "25.040.36223~e888ad7",
        "mt7981-wo-firmware": "20241110-r1",
        "default-settings": "29",
        "default-settings-chn": "29",
        "libuclient": "2024.10.22~88ae8f20-r1",
        "luci-i18n-attendedsysupgrade-zh-cn": "25.040.36223~e888ad7",
        "kmod-mt7981-firmware": "6.6.73.2025.01.14~8e4f72b6-r1",
        "kmod-lib-lzo": "6.6.73-r1",
        "kernel": "6.6.73~5dc876ca3b3685f0643c2f2c902f7b0b-r1",
        "luci-i18n-upnp-zh-cn": "25.040.36223~e888ad7",
        "acme-acmesh-dnsapi": "3.0.7-r2",
        "kmod-mppe": "6.6.73-r1",
        "hostapd-common": "2024.09.15~5ace39b0-r2",
        "kmod-crypto-arc4": "6.6.73-r1",
        "urngd": "2023.11.01~44365eb1-r1",
        "luci-i18n-firewall-zh-cn": "25.040.36223~e888ad7",
        "ppp-mod-pppoe": "2.5.1-r1",
        "dnsmasq-full": "2.90-r4"
    },
    "profile": "xiaomi,mi-router-ax3000t",
    "target": "mediatek/filogic",
    "version": "24.10.0",
    "diff_packages": true,
    "filesystem": "squashfs"
}
```

STDERR：
```
Package list missing or not up-to-date, generating it.

Building package index...
Downloading http://192.168.49.10/releases/24.10.0/targets/mediatek/filogic/packages/Packages.gz
Updated list of available packages in /builder/build_dir/target-aarch64_cortex-a53_musl/root-mediatek/../../../../builder/dl/immortalwrt_core
Downloading http://192.168.49.10/releases/24.10.0/targets/mediatek/filogic/packages/Packages.sig
Signature check passed.
Downloading http://192.168.49.10/releases/24.10.0/packages/aarch64_cortex-a53/base/Packages.gz
Updated list of available packages in /builder/build_dir/target-aarch64_cortex-a53_musl/root-mediatek/../../../../builder/dl/immortalwrt_base
Downloading http://192.168.49.10/releases/24.10.0/packages/aarch64_cortex-a53/base/Packages.sig
Signature check passed.
Downloading http://192.168.49.10/releases/24.10.0/targets/mediatek/filogic/kmods/6.6.73-1-5dc876ca3b3685f0643c2f2c902f7b0b/Packages.gz
Updated list of available packages in /builder/build_dir/target-aarch64_cortex-a53_musl/root-mediatek/../../../../builder/dl/immortalwrt_kmods
Downloading http://192.168.49.10/releases/24.10.0/targets/mediatek/filogic/kmods/6.6.73-1-5dc876ca3b3685f0643c2f2c902f7b0b/Packages.sig
Signature check passed.
Downloading http://192.168.49.10/releases/24.10.0/packages/aarch64_cortex-a53/luci/Packages.gz
Updated list of available packages in /builder/build_dir/target-aarch64_cortex-a53_musl/root-mediatek/../../../../builder/dl/immortalwrt_luci
Downloading http://192.168.49.10/releases/24.10.0/packages/aarch64_cortex-a53/luci/Packages.sig
Signature check passed.
Downloading http://192.168.49.10/releases/24.10.0/packages/aarch64_cortex-a53/packages/Packages.gz
Updated list of available packages in /builder/build_dir/target-aarch64_cortex-a53_musl/root-mediatek/../../../../builder/dl/immortalwrt_packages
Downloading http://192.168.49.10/releases/24.10.0/packages/aarch64_cortex-a53/packages/Packages.sig
Signature check passed.
Downloading http://192.168.49.10/releases/24.10.0/packages/aarch64_cortex-a53/routing/Packages.gz
Updated list of available packages in /builder/build_dir/target-aarch64_cortex-a53_musl/root-mediatek/../../../../builder/dl/immortalwrt_routing
Downloading http://192.168.49.10/releases/24.10.0/packages/aarch64_cortex-a53/routing/Packages.sig
Signature check passed.
Downloading http://192.168.49.10/releases/24.10.0/packages/aarch64_cortex-a53/telephony/Packages.gz
Updated list of available packages in /builder/build_dir/target-aarch64_cortex-a53_musl/root-mediatek/../../../../builder/dl/immortalwrt_telephony
Downloading http://192.168.49.10/releases/24.10.0/packages/aarch64_cortex-a53/telephony/Packages.sig
Signature check passed.
Downloading file:packages/Packages
Updated list of available packages in /builder/build_dir/target-aarch64_cortex-a53_musl/root-mediatek/../../../../builder/dl/imagebuilder
Downloading file:packages/Packages.sig
Signature check passed.
grep: /etc/nginx/module.d: No such file or directory
Collected errors:
 * check_data_file_clashes: Package luci-app-openvpn-server wants to install file /builder/build_dir/target-aarch64_cortex-a53_musl/root-mediatek/etc/config/openvpn
	But that file is already provided by package  * openvpn-openssl
 * opkg_install_cmd: Cannot install package luci-i18n-openvpn-server-zh-cn.
make[2]: *** [Makefile:234: package_install] Error 255
make[1]: *** [Makefile:171: _call_manifest] Error 2
make: *** [Makefile:349: manifest] Error 2
```

1715173329, 2025-04-28:
这状况大概只能再复制一份 init，连 config 一起重命名成 openvpn-server，甚至没办法做 config migration。

ali2liu, 2026-03-31:
这个问题要怎么才能解决呢？