immortalwrt/luci #579: luci-app-ipsec-vpnd: 安装报错

Saxon-Sun, 2025-08-17:
### Is there an existing issue for this?

- [x] I have searched among all existing issues (including closed issues)

### Actual behaviour

Collected errors:
 * opkg_conf_load: Could not lock /var/lock/opkg.lock: Resource temporarily unavailable.
WARNING: Strongswan is deprecating the ipsec CLI; please migrate to swanctl.
Command failed: ubus call service delete { "name": "ipsec-vpnd" } (Not found)
Command failed: ubus call service set { } (Invalid argument)

### Expected behaviour

.

### Steps to reproduce

系统→软件包→安装luci-app-ipsec-vpnd

### Additional Information

```shell
{
        "kernel": "6.6.93",
        "hostname": "ImmortalWrt",
        "rootfs_type": "squashfs",
        "release": {
                "distribution": "ImmortalWrt",
                "version": "24.10.2",
                "revision": "r33247-467867283bb9",
                "target": "x86/64",
                "description": "ImmortalWrt 24.10.2 r33247-467867283bb9",
                "builddate": "1750865788"
        }
}
```

### What browsers do you see the problem on?

Chrome

### Relevant log output

```shell

```

Ansen, 2025-08-27:
Could not lock /var/lock/opkg.lock:
应该是还在运行安装其它软件

Saxon-Sun, 2025-08-27:
> Could not lock /var/lock/opkg.lock: 应该是正在安装其它软件

但是就算是编译出来的ipsec也开不起来，你可以试一下 😂

Saxon-Sun, 2025-08-27:
> > > 建议用 softethervpn
> > 
> > 
> > 也启动不了啊 天灵大佬 勾选了保存后一直显示未运行
> 
> 日志呢？

怎么看日志呀 我在luci的系统日志里看不到关于softethervpn的错误日志 ，好像就像是没启动一样 但是勾选后保存后，手动去启动里面启用softethervpn，softethervpn页面又显示运行了，奇奇怪怪 你们的24.10都能正常勾选保存打开吗？🥹

Ansen, 2025-08-28:
> > 建议用 softethervpn
> 
> 也启动不了啊 天灵大佬 勾选了保存后一直显示未运行

显示未运行, 实际是可以连接的, 它这里的显示有点问题
我看源码它获取的是 [ipsec-vpnd 的状态](https://github.com/immortalwrt/luci/blob/master/applications/luci-app-ipsec-vpnd/htdocs/luci-static/resources/view/ipsec-vpnd.js#L20),  我手动修改成获取 ipsec 的状态就行了

因为我注意到 lean大的版本中, ipsec-vpnd 也是不运行的, 但是它是通过 [pgrep ipsec 来获取状态](https://github.com/coolsnowwolf/luci/blob/master/applications/luci-app-ipsec-vpnd/luasrc/controller/ipsec-server.lua#L15)

PS:  个人觉得 softethervpn 用起来挺麻烦的,还需要安装管理端, 习惯用 luci-app-ipsec-vpnd 了

Saxon-Sun, 2025-08-28:
> > > Could not lock /var/lock/opkg.lock: 应该是正在安装其它软件
> > 
> > 
> > 但是就算是编译出来的ipsec也开不起来，你可以试一下 😂
> 
> 显示未运行, 实际是可以连接的, 它这里的显示有点问题
> 
> 我看源码它获取的是 [ipsec-vpnd 的状态](https://github.com/immortalwrt/luci/blob/master/applications/luci-app-ipsec-vpnd/htdocs/luci-static/resources/view/ipsec-vpnd.js#L20), 我手动修改成获取 ipsec 的状态就行了, 因为时间有限(懒), 就没去通读源码搞懂为何使用ipsec-vpnd了
> 
> 因为我注意到 lean大的版本中, ipsec-vpnd 也是不运行的, 只会运行 ipsec 服务, 但是lean是通过 [pgrep ipsec 来获取状态](https://github.com/coolsnowwolf/luci/blob/master/applications/luci-app-ipsec-vpnd/luasrc/controller/ipsec-server.lua#L15)
> 
> PS: 个人觉得 softethervpn 用起来挺麻烦的,还需要安装管理端, 习惯用 luci-app-ipsec-vpnd 了

感觉不止是状态获取的问题，我启动开关好像没啥用，我关了开关后还能链接，我在终端把ipsec进程kill了后还会自动启动，人都整蒙了

Saxon-Sun, 2025-08-28:
> > > > > Could not lock /var/lock/opkg.lock: 应该是正在安装其它软件
> > > > 
> > > > 
> > > > 但是就算是编译出来的ipsec也开不起来，你可以试一下 😂
> > > 
> > > 
> > > 显示未运行, 实际是可以连接的, 它这里的显示有点问题
> > > 我看源码它获取的是 [ipsec-vpnd 的状态](https://github.com/immortalwrt/luci/blob/master/applications/luci-app-ipsec-vpnd/htdocs/luci-static/resources/view/ipsec-vpnd.js#L20), 我手动修改成获取 ipsec 的状态就行了, 因为时间有限(懒), 就没去通读源码搞懂为何使用ipsec-vpnd了
> > > 因为我注意到 lean大的版本中, ipsec-vpnd 也是不运行的, 只会运行 ipsec 服务, 但是lean是通过 [pgrep ipsec 来获取状态](https://github.com/coolsnowwolf/luci/blob/master/applications/luci-app-ipsec-vpnd/luasrc/controller/ipsec-server.lua#L15)
> > > PS: 个人觉得 softethervpn 用起来挺麻烦的,还需要安装管理端, 习惯用 luci-app-ipsec-vpnd 了
> > 
> > 
> > 显示未启动，但是要是在luci-启动里重启ipsec和ipsec-vpn，立马能用了，也显示运行中了，softethervpn也是这样😅
> 
> 将就用吧, 看作者大大的意思是不想再为这东西花费精力了

等有缘人pr修复了🥹

kissingers, 2026-03-14:
一样，最近编译的这个模块太折腾人了，有时候能起来，有时候不能起来。