immortalwrt/luci #594: luci-app-smartdns: 开启SmartDNS时，会导致Dnsmasq下发DNS错误

ListenV, 2025-09-28:
### Is there an existing issue for this?

- [x] I have searched among all existing issues (including closed issues)

### Actual behaviour

我使用的是LibWrt，基于Immortalwrt。Immortalwrt下我没有测试。

我不清楚这个问题是否应该给 [pymumu/luci-app-smartdns](https://github.com/pymumu/luci-app-smartdns) 发 issue，如果有需要，我可以去发。

luci-app-smartdns 版本：46

我的内网IP：`network.lan.ipaddr='10.1.1.1/24'`（我试过将其设置为`10.1.1.1`，不会正常工作）

开启SmartDNS时，自动添加一条`dhcp_option`，会直接将上面的值`10.1.1.1/24`拿来用，会设置成 `dhcp.lan.dhcp_option='6,10.1.1.1/24'`，这会导致dnsmasq无法正确识别，会给客户端分配错误的dns `24.10.1.1`。正确的值应该是 `dhcp.lan.dhcp_option='6,10.1.1.1'`。

### Expected behaviour

期望正确设置`dhcp_option`

### Steps to reproduce

直接开启SmartDNS就会复现

### Additional Information

```shell
{
        "kernel": "6.12.45",
        "hostname": "Router",
        "system": "ARMv8 Processor rev 4",
        "model": "JDCloud AX1800 Pro",
        "board_name": "jdcloud,re-ss-01",
        "rootfs_type": "erofs",
        "release": {
                "distribution": "LibWrt",
                "version": "SNAPSHOT",
                "firmware_url": "https://downloads.immortalwrt.org/",
                "revision": "r0-200988c",
                "target": "qualcommax/ipq60xx",
                "description": "LibWrt SNAPSHOT r0-200988c",
                "builddate": "1758153088"
        }
}
```

### Relevant log output

```shell
没有发现dnsmasq的异常日志
```

xlighting2017, 2025-09-29:
https://github.com/pymumu/smartdns/commit/78aa7ea83891d04237a7567961b80bc575422ab4

对应版本是 47

1715173329, 2025-09-29:
第三方 fork 不受本项目支持。